SteveMacenski/spatio_temporal_voxel_layer #311: Multiple Pointcloud2 subscribers with fast dds

luke22zzang, 2024-12-26:
I am working with two RealSense D435 cameras to apply PointCloud data to the Spatio-Temporal Voxel Layer (STVL). However, when RViz or rosbag simultaneously subscribes to the PointCloud topics, the callback in STVL is not triggered.

Previously, I found that this issue(https://github.com/ros-navigation/navigation2/issues/3014) was caused by Fast DDS, and switching to Cyclone DDS resolved the problem. Later, it seemed that this issue was fixed(https://github.com/ros2/rmw_fastrtps/pull/619) in newer versions of Fast DDS. However, the issue still persists when using Fast DDS.

Could there be another underlying cause for this?
Why does this problem occur specifically with Fast DDS?
I want to understand the root cause. Please help me.

Thank you.

OS: ubuntu 22.04
ROS: humble
DDS: Fast DDS
Camera: Realsense D435 * 2
QoS: Sensor QoS

[image]

SteveMacenski, 2025-01-02:
Generally, this isn't an issue with STVL so I'm going to close this. It is unfortunate that you're running into issues - perhaps caused by Fast DDS - but there's nothing actionable at the `rclcpp` or application layer (STVL) to be done about this. 

Have you tried reporting to Fast-DDS / RMW? 

Though a quick question: is rviz running on a remote machine or the robot itself? It could just be that you're dropping packets over the wifi from the robot to your development machine and it is not representative of on-CPU performance (or that STVL isn't properly publishing within the robot).

closed 2025-01-02 by SteveMacenski

luke22zzang, 2025-01-09:
Thank you. I will try reporting to fast dds.

All are working in local pc.