SteveMacenski/spatio_temporal_voxel_layer #313: find_package() error

Alessio-Lovato-Unipd, 2025-02-10:
Hello, I am currently using ROS2 Humble on Ubuntu 22.04.4.
I have installed the package through apt 
`sudo apt-get install ros-humble-spatio-temporal-voxel-layer`
I have STVL as dependency to another package using the command
`find_package(spatio_temporal_voxel_layer REQUIRED)`
but during compiling I obtain the following error:
```
CMake Error at /opt/ros/humble/share/spatio_temporal_voxel_layer/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'spatio_temporal_voxel_layer' exports the library
  'spatio_temporal_voxel_layer' which couldn't be found
Call Stack (most recent call first):
  /opt/ros/humble/share/spatio_temporal_voxel_layer/cmake/spatio_temporal_voxel_layerConfig.cmake:41 (include)
  CMakeLists.txt:18 (find_package)
```

I don't know much about cmake, but I suspect the problem might be the ament_export_library which [CMakeLists.txt](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/ros2/spatio_temporal_voxel_layer/CMakeLists.txt) at line 136, which exports libraries with the name of the package but all the libraries are built using the 'library_name' variable.

SteveMacenski, 2025-02-10:
I agree that's probably the error. Can you change that to the library export and see if that resolves your issue? If so, please open a PR and I can work on a backport to Humble this week so its out next time there's a sync!

AlessioLovato, 2025-02-11:
@SteveMacenski, I can confirm that was indeed the problem. I was able to successfully build the STVL package from source (I downloaded openvdb-vendor from apt as it was too large to build from source) and find it with find_package().

One thing I haven't quite understood: should I submit a PR to the humble or the ros2 branch? (FYI, I'm unable to test the ros2 branch.)

SteveMacenski, 2025-02-11:
`ros2` at least so that all new distributions have this fix, but also humble / jazzy if you want it in either of those :-) Its a one-liner change so easy enough to do in the github UI