SteveMacenski/spatio_temporal_voxel_layer #319: STVL only represent obstacles in a single interval?

Baominchao, 2025-03-13:
I use STVL to detect the height range of -1.0 to 1.5, and then -0.1 to 0.2 is the normal walking area, while [-1.0, -0.1] [0.2, 1.5] are other obstacles. Is that okay? May I set it up like this? Or can STVL only represent obstacles in a single interval?Because the scenarios used include water channels below the walking plane, etc.Looking forward to your response.

SteveMacenski, 2025-03-13:
I don't quite understand. `-0.1 - 0.2` is within the bounds of `-1.0 - 1.5` so that is already included, no additional setup is required.

Baominchao, 2025-03-14:
I'm very sorry I didn't express myself clearly. This problem can be described as follows:
I have set the following parameters:
min_obstacle_height: -1.0
max_obstacle_height: 1.5
Setting min_obstacle_height to a negative number is intended to be lower than the ground, such as a canal in farmland, which is recognized as an obstacle. If set in this way, will the ground on which the robot walks (height=0.0) be recognized as an obstacle?

SteveMacenski, 2025-03-14:
> will the ground on which the robot walks (height=0.0) be recognized as an obstacle?

Yes. You could conceptually have multiple instances of the layer through that address different settings if you want to have multiple ranges (10cm - 1.5m; -1.0m - -10cm; ignoring -10cm - 10cm) which is a not-so-uncommon setup if you have multiple sensors and those sensors are in vastly different height ranges on the robot and/or application

Baominchao, 2025-03-16:
Thank you very much for your reply. I have gained a clearer understanding of this issue. Wishing you a happy life and all the best.

closed 2025-03-16 by SteveMacenski