SteveMacenski/spatio_temporal_voxel_layer #320: Voxels outside the costmap are not cleared when using clearExceptRegion or clearAroundRobot

tonynajjar, 2025-03-18:
This issue could also be in Nav2 since I believe it's an interface issue.

With clearExceptRegion and clearAroundRobot, [we enforce the start and end point to be in the costmap](https://github.com/ros-navigation/navigation2/blob/c7a5cc56f6b460a818cad27c5c47848f0ab9f0d0/nav2_costmap_2d/src/clear_costmap_service.cpp#L138)

So in STVL's `clearArea`, the voxels outside the costmap are not cleared

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/7673fac40bb41ebf7419166045442a9d0dcd8af5/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_layer.cpp#L968

With `clearEntire`, it is not a problem because we call `resetLayers()` which resets the whole grid [here](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/7673fac40bb41ebf7419166045442a9d0dcd8af5/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_layer.cpp#L603)

tonynajjar, 2025-03-18:
Right now because of this bug https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/321, the described issue happens currently with `ClearCostmapExceptRegion` instead of `clearAroundRobot`

SteveMacenski, 2025-03-18:
Just to be clear, this is still an issue not solved in #321 ? 

If they're outside of the non-rolling costmap, then it doesn't really matter if we clear or not any STVL voxels since they would never appear in projections within the costmap's bounds anyway to impact navigation. It might have some compute overhead until they expire, but that seems relatively minimal.

For the rolling case, if we're clearing everything outside of the radius around the robot, this works properly, no? We clear everything outside of those bounds, regardless of costmap size by iterating over the whole grid. If we're clearing everything inside, I'm not understanding the issue if the bounds are the same in STVL's grid as is the costmap layer, as is other costmap layers -- its consistent, no?

tonynajjar, 2025-03-18:
> Just to be clear, this is still an issue not solved in #321 ?

Yes it's different

> For the rolling case, if we're clearing everything outside of the radius around the robot, this works properly, no? We clear everything outside of those bounds, regardless of costmap size by iterating over the whole grid

Works as long as the radius is smaller than the costmap size. The radius is limited to the costmap size but I'd like it to be bigger.

> It might have some compute overhead until they expire, but that seems relatively minimal.

In my use case, decay time is huge, basically permanent (because of limited perception FOV). So over time more and more voxels accumulate and CPU usage increases.

SteveMacenski, 2025-03-19:
Oh.. your radius is set generally larger than your local costmap size? That is an interesting case I hadn't considered. 

That seems more a question for Nav2 than STVL, no? We could check in Costmap2D if that's the case and clear the local costmap completely since that should be an eq. operation in that case. That should handle your comments in STVL implicitly

tonynajjar, 2025-03-19:
Turns out radius = costmap size is fine for me now so I can't justify spending more time on this right now. 

Just for tracking, I tried moving the bounding of the coordinates to after STVL's `clearArea` but that didn't work and I'm not sure why -> calling clearExcept with a radius bigger than the costmap size was just clearing everything

you can see it here: https://github.com/angsa-robotics/navigation2/compare/jazzy...angsa-robotics:navigation2:clear-area-fix?expand=1