SteveMacenski/spatio_temporal_voxel_layer #323: Feature request: Clear voxels based on distance

tonynajjar, 2025-03-28:
In addition to clearing voxels based on time, I'd also like to clear voxels based on distance

This is motivated by my following use-case: I have a narrow obstacle-detection FOV which requires the voxel decay to be very high (basically permanent) to be on the safe side and not temporaly clear voxels out of the FOV. The voxels might never be in the FOV again (e.g. think of a robot passing a narrow corridor with a narrow front-facing FOV) but need to be cleared eventually for CPU/performance reasons. The safest way I can think of is based on distance ->  if the robot is too far away, clear the voxel.

tonynajjar, 2025-03-28:
I took shot at it and tested it, it works well: https://github.com/angsa-robotics/spatio_temporal_voxel_layer/pull/1
I'd be happy if someone has the time to refine it enough so we can merge it in this repo. Some things to rethink/generalize/improve:

- for my application I only care about the 2d distance so that's how I implemented it
- is it fine/clean to pass the tf buffer to the SpatioTemporalVoxelGrid?
- documentation

SteveMacenski, 2025-04-01:
> for my application I only care about the 2d distance so that's how I implemented it

I think that's sensible

> is it fine/clean to pass the tf buffer to the SpatioTemporalVoxelGrid?

I don't like that. I also see that you look up transforms in the clearing loop which has _got_ to be computationally slow. I think you should just be able to get the robot's pose in the grid frame once, pass that into the grid for the clearing operation, and compare that to the voxel location instead.

I'd also check the `dist_2` rather than `dist` to above the `sqrt` operation which is costly in high volumes

> for semantic reasons maybe the spatial clearing should be taken out of TemporalClearAndGenerateCostmap or have the function renamed

Perhaps!

---

I will say that the intent of the global decay time is to remove very old, distance voxels from circulation. Basing it on distance too to forget is sensible, but also if you set the global decay to ~1-2 minute, that seems long enough that you'll have long since left that area even at a slow speed like 0.5 m/s

tonynajjar, 2025-04-01:
> but also if you set the global decay to ~1-2 minute, that seems long enough that you'll have long since left that area even at a slow speed like 0.5 m/s

I can't set it to 1-2 mins because I can't assume that the robot is always moving because the mission can be paused for an extended amount of time. It should therefore keep track of all voxels permanently-ish

SteveMacenski, 2025-04-01:
Ok, I like this feature then - with those performance changes, I'd be down with merging this into mainline!

closed 2025-04-02 by SteveMacenski